DroidPlanner/Tower #1518: Add support for Joystick control

ne0fhyk, 2015-07-14:
Will help with development for the ArduCopter on bebop!

ne0fhyk, 2015-08-05:
@TShapinsky dependent on tests results for the [`SET_POSITION_TARGET_GLOBAL_INT`](https://pixhawk.ethz.ch/mavlink/#SET_POSITION_TARGET_GLOBAL_INT) mavlink message.

chocol4te, 2016-05-07:
Has any progress been made on this? I am trying to compile the virtual_joystick branch and so far with no luck. Has anyone managed to build an APK?

Guiboy, 2016-09-14:
Would be as well very much interested in having virtual joystick support in Tower. This function already is implemented in other GCS's as mission planner for example (and several others)
This is a real pity as I consider Tower as the best user-friendly-robust GCS in the landscape today.
Does anybody plan to release a beta for testing? I am up with it!
Guillaume

Mocart96, 2017-10-04:
Can you say little more how it will be working ? If user enable virtual joystick will it be possible to arm FC without R/C?

TShapinsky, 2017-10-05:
At the time of my first implementation the virtual joystick was not piped over rc overrides so you would not be able to arm with them.