DroidPlanner/Tower #1524: After resuming paused auto-mission RIO is lost

TShapinsky, 2015-07-21:
there is a command `MAV_CMD_OVERRIDE_GOTO` that would tell the copter to maintain current position, the problem is that it isn't implemented by anything.

squilter, 2015-07-21:
It's not implemented in ArduCopter as far as I can tell.  But supposedly mission planner has the ability to pause a mission ("supposedly"), so let's try looking into how that was implemented.

hamishwillee, 2015-07-21:
Probably not helpful - but I *think*if  you switch out of auto the mission is paused and the copter loiters. There is a parameter that you can set to determine whether the mission restarts from where it left off or the beginning again.

ne0fhyk, 2015-09-14:
Closing until support is added from the autopilot.

closed 2015-09-14 by ne0fhyk