DroidPlanner/Tower #1556: AC 3.3 App crashes when Brake-mode is issued

deepthought-64, 2015-08-03:
When connected to the copter the app crashes as soon as the brake-mode is activated over the RC-TX via a CHx_OPT switch. 

Cause: 
`IllegalArgumentException. COPTER_BREAK is not a constant in com.o3dr.services.android.lib.drone.property.VehicleMode`

Seems like the version I use (Tower-v3.1.6) does not know about the break mode of AC 3.3 (I use -rc8 at the moment) yet.

How to reproduce:
* Assign brake mode to a switch on the TX
* Connect the smartphone/tablet to the copter either via RF-link or USB-Cable
* Start the Tower App and connect to the copter. 
* Switch on the brake-mode --> App crashes

ne0fhyk, 2015-08-03:
Thanks for reporting! That issue has already been fixed and will be in the next release.

closed 2015-08-03 by ne0fhyk

deepthought-64, 2015-08-03:
Great to hear that!