DroidPlanner/Tower #1559: Tower crashes when drawing survey region

franknitty69, 2015-08-07:
Go to editor
select spline waypoint>survey
draw your survey region
when the region is completed, tower crashes with a message "Unfortunately, Tower has stopped".

this happens 100% of the time. i have tried rebooting and a re-installation of Tower.

Tower: v3.1.6
3DR Services: v1.3.2
Device: Samsung Galaxy Tab S 8.4

ne0fhyk, 2015-08-07:
@franknitty69 thanks for reporting. We'll look into the issue.

ne0fhyk, 2015-08-07:
@franknitty69 the issue is caused by the beta version of 3DR Services you're running. 
A fix will be released with 3DR Services v1.3.3.

ne0fhyk, 2015-08-07:
Fixed by commit https://github.com/dronekit/dronekit-android/commit/2d469ce21de2514486cd5a686bdec578b36be2da

closed 2015-08-07 by ne0fhyk