DroidPlanner/Tower #1563: Battery selector

ThorstenDP, 2015-08-11:
Hi all,

there is an increasing number of battery specific parameter settings:
- BATT_CAPACITY
- FS_BATT_MAH
- FS_BATT_VOLTAGE
- MOT_THST_BAT_MAX
- MOT_THST_BAT_MIN
When changing payloads or when traveling it is often required to change the batteries.
Scanning though the full parameter list and setting all parameters one by one is cumbersome and can easily lead to errors.
Hence, I suggest to include some option (maybe by using a specific xml file for each battery) to easily switch between batteries.

Thanks and best regards,
Thorsten

squilter, 2015-08-11:
I think the easiest way to do it right now is to save all of your settings to a txt file.  Save a file for each battery.  Then just upload the right one.

ThorstenDP, 2015-08-11:
Hi Sebastian,
yes, that's true and definitely an option for now. 
But I am concerned - especially if you use a rather slow telemetry like xBee - that not all parameters updates are checked (I as far as I know only happens for missions). Hence, you always have to download the parameters again to check if the changes were really uploaded. 
Another problems is that if you change some other parameters you have to change it in all files.
Cheers,
Thorsten