DroidPlanner/Tower #1572: Tower changing flight mode by itself

lvale, 2015-08-21:
Hi,

I was hovering a small quad at low height to estimate battery, i.e. waiting for the battery failsafe to kick in and do the LAND action. I had PreArm checks disabled because I was doing this indoors.

For the first time I've used Tower to periodically tell me the batt voltage, nothing else, i.e. no actions performed from the app.

When the battery reached the correct value the quad landed (Tower said LAND) and just before touching the ground I heard Tower say Loiter. 

To my surprise and without any user intervention Tower decided to switch mode to Loiter, even if I was without GPS, and even allowed me to arm in Loiter.

Please stop this action because it might be dangerous. I have logs from the copter, and might be able to recover the blog from the Android tablet.

jmachuca77, 2015-08-21:
Hi Luis, I don’t think tower can change the mode to loiter if there is no GPS, ardupilot should reject the request if there is no GPS I would suggest you post the logs.

lvale, 2015-08-21:
Gracias Jaime.

That was my impression also. I'm now looking at a huge log, but even if the copter had perfect GPS coverage it should not change the mode from LAND to LOITER.

Both Mission Planner and APM Planner don't behave like this, although with specific quirks...ie if we try to arm without noticing that LAND was engaged both ask to check the safety switch...

lvale, 2015-08-21:
Olá Jaime

It did change to Loiter with GPS status of 1.....

Part of the log here:
https://drive.google.com/file/d/0BzRxu4IwHPYCZzNuU3ltYXk4YXc/view?usp=sharing

ne0fhyk, 2015-08-22:
@lvale Tower does change the mode to `loiter` on land. Here's the history behind this behavior: https://github.com/DroidPlanner/Tower/issues/1108

jmachuca77, 2015-08-22:
Ok, so I guess if its disarmed it would probably allow the change to loiter.

lvale, 2015-08-22:
@jmachuca77 Part of the issue: http://diydrones.com/forum/topics/copter-3-3-beta-testing?commentId=705844%3AComment%3A2075783&xg_source=msg_com_forum

ne0fhyk, 2015-08-23:
@lvale after reading the history behind it, is the current behavior still an issue for you?

rcglider, 2015-08-24:
I have had this happen every time IIRC using Land or RTL. Loiter is not even set as a flight mode in the tx.

ne0fhyk, 2015-08-24:
The behavior is occurring in 3DR Services, and I have created [an issue](https://github.com/dronekit/dronekit-android/issues/246) to have it fixed for the `v1.3.4` release.

To match the change in behavior, an option will be added to Tower's settings to disable the mode switch on disarm.

rmackay9, 2015-08-24:
Just to add my two cents, I think the GCS should avoid ever change the flight mode unless as it's a result of input from the user.  Otherwise it just creates confusion as to what the vehicle's flight mode is.

So for example, I think it's fine for the GCS to switch the vehicle to Guided when the user pushes the takeoff button (and this is what it should do).  I don't think it's ok for the GCS to switch the vehicle to Loiter when it thinks the vehicle has landed.  Really the GCS doesn't know what the user's preferred flight mode is and the user should remain in control of this.

lvale, 2015-08-24:
@ne0fhyk 
I believe the behavior of automating changes to the flight modes should be avoided as much as possible.

The switch to LAND and subsequent DISARM was caused by a vehicle Failsafe action, which is expected behavior. Also the control was never done by Tower but by the Radio. 

When the bug that Randy traced on the Loiter code then Tower could have tried to set Loiter unsuccessfully because the vehicle would reject it, in the case without GPS I was using, but, as Randy says, mode changes unless caused by user action or vehicle failsafe should be avoided.

cala2, 2015-08-24:
I experienced a change mode today with my airplane, what I did:I updated tower to beta and 3dr services but for safe (not experienced pilot and read about this issue) I return to stable vers from google play and go to fly, mission was in APM previously, I changed from stab to auto from the tower and start mission, in WP 5 the plane returned to stab alone, I advertise because I have FPV and osd that shows the change but Tower didn't say nothing on the audible mode; It's looks a risk issue if you are flying only with the tablet, If you think convenience I can advertise from this in Tower group until is solved.

ne0fhyk, 2015-09-14:
This issue was fixed in 3DR Services v1.3.4.

closed 2015-09-14 by ne0fhyk