DroidPlanner/Tower #1574: App crashes when attempting to connect with bluetooth but no devices paired

rmackay9, 2015-08-24:
The titles says it all but I started up Tower v3.1.6 with 3DR services v1.3.3 and attempted to connect using BlueTooth.  I had not yet paired it with a bluetooth devices so after pushing connect it popped up a window asking me to select a bluetooth device but there were none in the list.

here are links to the log files on the android device.  Looks like it's got something about the cause of the crash.
https://dl.dropboxusercontent.com/u/3852647/Randy-TowerBluetooth/2015-08-24%2011-50-37.txt
https://dl.dropboxusercontent.com/u/3852647/Randy-TowerBluetooth/2015-08-24%2011-51-50.txt

ne0fhyk, 2015-08-24:
Fixed by commit 7418f0adedd5685b0a7be7ceec1f079ce12a8980

closed 2015-08-24 by ne0fhyk

rmackay9, 2015-08-24:
great, thanks!