DroidPlanner/Tower #1575: Arming and Takeoff should only be from Guided mode

rmackay9, 2015-08-24:
At least with AC3.3 I think that tower should only allow arming and take-off from Guided mode.

I don't think there's any point in allowing the user to use the GCS to arm and then take-off in other modes like Loiter mode because it means the user's fingers are not on the TX so it's slightly dangerous.

Solo's controller is an exception to this because the take-off command is sent from the TX's take-off button so the user's fingers are on the TX when the take-off is issued.

ne0fhyk, 2015-09-14:
@rmackay9 Tower allows `arming` from any modes that supports it as long as the arming conditions are met, but `takeoff` is only done in `guided` mode. 
Does that resolve the issues you're thinking of?

ne0fhyk, 2015-09-26:
@rmackay9 any comments on this issue?