DroidPlanner/Tower #1581: smoother Copter follow-me with SET_POSITION_TARGET_GLOBAL_INT

rmackay9, 2015-08-28:
The current follow-me code sends a MISSION_ITEM or COMMAND_LONG with a MAV_CMD of WAYPOINT within it to command the vehicle to move to a particular location.  A smoother and more accurate control of the vehicle is possible if using Copter-3.3 if the SET_POSITION_TARGET_GLOBAL_INT command was used instead.

The reason it would be more accurate is that the target location is specified as an integer which allows 1cm precision while the current method uses a float which is accurate to only 1m (although it depends upon your location in the world, when closer to 0,0 it actually becomes more accurate)

http://mavlink.org/messages/common#SET_POSITION_TARGET_GLOBAL_INT

ne0fhyk, 2015-08-31:
@rmackay9 [3DR Services](https://github.com/dronekit/dronekit-android/blob/e77f48588aec1f1cca014ec23b23ff89c6576f36/ServiceApp/src/org/droidplanner/services/android/core/MAVLink/MavLinkModes.java) already has support for this, but it was disabled until AC 3.3 was released.

I'll update the issue's title to reflect the work remaining.

ne0fhyk, 2015-08-31:
Replaced by issue https://github.com/dronekit/dronekit-android/issues/248

closed 2015-08-31 by ne0fhyk