DroidPlanner/Tower #1589: Tower crashes when switching Map Tile Provider

franknitty69, 2015-09-09:
[B]steps to reproduce[/B]
-go to settings>user interface>map provider preferences
-press Map Tile Provider, in the modal window, select the Mapbox radio button
-after a few seconds, tower crashes with notification that says "Unfortunately, Tower has stopped". The notification has Report or OK buttons.
-press ok to dismiss the notification and tower is restarted back in map provider preferences

[B]expected result[/B]
the map provider preferences is changed to mapbox and mapbox preferences are enabled

ne0fhyk, 2015-09-09:
Which version of the tower app are you seeing the crash on?

franknitty69, 2015-09-10:
Tower: v3.2.0.beta.3
3DR Services: v1.3.4

guiseco, 2015-09-11:
09-11 10:38:06.181 23281 23281 E AndroidRuntime: FATAL EXCEPTION: main

09-11 10:38:06.181 23281 23281 E AndroidRuntime: Process: org.droidplanner.android, PID: 23281

09-11 10:38:06.181 23281 23281 E AndroidRuntime: java.lang.IllegalStateException: Parent activity must implement org.droidplanner.android.dialogs.EditInputDialog$Listener

09-11 10:38:06.181 23281 23281 E AndroidRuntime: 	at org.droidplanner.android.dialogs.EditInputDialog.onAttach(EditInputDialog.java:68)

09-11 10:38:06.181 23281 23281 E AndroidRuntime: 	at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:853)

09-11 10:38:06.181 23281 23281 E AndroidRuntime: 	at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1067)

09-11 10:38:06.181 23281 23281 E AndroidRuntime: 	at android.app.BackStackRecord.run(BackStackRecord.java:833)

09-11 10:38:06.181 23281 23281 E AndroidRuntime: 	at android.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:1452)

09-11 10:38:06.181 23281 23281 E AndroidRuntime: 	at android.app.FragmentManagerImpl$1.run(FragmentManager.java:447)

09-11 10:38:06.181 23281 23281 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:739)

09-11 10:38:06.181 23281 23281 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:95)

09-11 10:38:06.181 23281 23281 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:135)

09-11 10:38:06.181 23281 23281 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:5253)

09-11 10:38:06.181 23281 23281 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)

09-11 10:38:06.181 23281 23281 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Method.java:372)

09-11 10:38:06.181 23281 23281 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:899)

09-11 10:38:06.181 23281 23281 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:694)

09-11 10:38:06.181   533  1204 W ActivityManager:   Force finishing activity org.droidplanner.android/.activities.helpers.MapPreferencesActivity

ne0fhyk, 2015-09-14:
Fixed by pr #1596

closed 2015-09-14 by ne0fhyk