DroidPlanner/Tower #1590: Tower causing device crash and reset moments after connecting via USB 3dr radio

Jimmydaux, 2015-09-10:
I have a Samsung Galaxy S6 that crashes every time I connect tower to my multirotors. This happens consistently with two different vehicles, two different radios and two different OTG cables. My nexus 7 does not crash, but my S6 crashes every time. It will not crash with the OTG and radio plugged in alone, but crashes moments after hitting the connect button.

Jimmydaux, 2015-09-28:
Fyi, I tried this on a second samsung galaxy s6 and had the same behavior. My wife's phone crashes and resets just as mine does.

ne0fhyk, 2015-10-02:
@Jimmydaux I was not able to reproduce the issue on a galaxy S6 device. What version of 3DR Services and Tower are you running?

Jimmydaux, 2015-10-03:
@ne0fhyk THank you for taking a look at this. 

My current 3dr Services is v1.4.2.beta.1 and Tower is v3.2.0.beta.6 and my quad is currently on AC3.3 rc11. I have tried pretty much every version of tower and 3dr services for the last couple of months hoping that something would have changed enough to solve the issue, but to no avail. I even dropped back to droidplanner and it did it there as well. 

Both S6's that have the issues are verizon, if that somehow affects something. Should I take a video of it forcing a restart or is there any other information that could be helpful in trying to figure out why this happens?

Jeremy

ne0fhyk, 2015-10-03:
@Jimmydaux There should be log files for 3DR Services in the `/sdcard/Android/data/org.droidplanner.services.android/files/log_cat/` directory. Could you replicate a crash, and share the resulting log file (on dropbox or google drive).

Jimmydaux, 2015-10-03:
Here is a link to a log i just barely generated. Just to provide some context on what you will see, I connected via 3DR radios and then disconnected immediately after. This didnt crash my phone. I reconnected and waited to see if the first connect and reconnect would help.... It didnt. It crashed and restarted once again 

http://1drv.ms/1j6w7wc

ne0fhyk, 2015-10-03:
@Jimmydaux thanks for the logs. I'm not seeing anything out of the ordinary from these.
Could you share the Tower logs as well. These should be in the `/sdcard/Android/data/org.droidplanner.android/files/crash_log/` directory.

By the way, when you said it `crashes your phone`, do you mean `Tower` crashed and restarted, or that your phone rebooted upon connection?

Jimmydaux, 2015-10-04:
@ne0fhyk 

The tower log from the same "crash" is here:

http://1drv.ms/1Z16j4O

When Tower crashes my phone, it goes from being connected and communicating with my quad (5-10 seconds tops) straight to a black screen and the phone itself is forced to restart. There is no in-between where it looks like tower shuts down, it just goes from being connected for a few seconds to full device reboot.

Jimmydaux, 2015-10-06:
Is there a way for me to pull and share an android crash log to see if there is any info there that could be helpful?

js349, 2015-10-09:
Tower also crashes my Note 5 (connects to drone then reboots phone within a few seconds) which is on Android 5.1.1. I have the latest version of Tower and 3DR installed on my phone.

lvale, 2015-10-09:
Same here, both on OnePlus One with Lollipop and Toshiba AT300E tablet with 4.1.1.

I'm uploading logs now to GDrive

js349, 2015-10-09:
They (Tower and 3DR) work fine on my Samsung Galaxy Tab 3 with 4.2.

Jimmydaux, 2015-10-10:
@ne0fhyk I forgot to tag you in my last post. Is there anything that I can get to you from android logs that would help you figure out what is going on?

Thoin, 2015-10-11:
I have 2 different Samsung devices that show this issue. I will first explain the issue, then describe both devices and installed versions. I will attach the appropriate logs to this post as well.

**The Issue**: When I connect the genuine 3DR radio to my devices with the provided USB OTG cable, and click connect, Tower crashes instantly, followed shortly after (less than 5 seconds) by the 3DR Services app crashing.

**Situation**: Pixhawk is powered on and connected to paired radio. I have tested with QGroundControl on my PC and the radios are functioning correctly. Radio on android device has power and is communicating with android as the USB interact lights flash inside the radio and the green light is flashing.

**Devices**: Both devices crash in the very same fashion.
*Device 1*: Samsung Galaxy S5
OS: Stock Android 5.0
3DR Services version 1.4.2
Tower version 3.2.0

*Device 2*: Samsung Galaxy Tab S
OS: Stock Android 5.0.2
3DR Services version 1.4.2
Tower version 3.1.6

Device|Tower Logs|3DR Service Logs
---------|---------------|-----------------------
Galaxy S5|https://drive.google.com/folderview?id=0B-4Pt7stMkc8LUZoZmNhVW9GeXc&usp=sharing|https://drive.google.com/folderview?id=0B-4Pt7stMkc8X0VCcG9rOWp1TlU&usp=sharing
Galaxy Tab S|https://drive.google.com/folderview?id=0B-4Pt7stMkc8bFFZTzZOV0JJakE&usp=sharing|https://drive.google.com/folderview?id=0B-4Pt7stMkc8enduVlhKejg0MFk&usp=sharing

[All Crash Logs for Download in a Single Folder](https://drive.google.com/folderview?id=0B-4Pt7stMkc8ZVBRUnJ3aFBSYUk&usp=sharing)

ne0fhyk, 2015-10-11:
@lvale I'm unable to access the logs. It says the items no longer exists.

lvale, 2015-10-11:
Sorry, forgot I had a local sync and was moving directories around....

http://1drv.ms/1hywkqa

Jimmydaux, 2015-10-21:
Is there anything else that I can provide to help resolve this crashing issue?

ontogmolomo, 2015-10-21:
good day. i managed to solved the issue by uninstalling ang installing the
app.

thank you for the support.

Jimmydaux, 2015-10-23:
I just wanted to be sure that it is known that, at least for me, uninstalling and reinstalling did not solve my issue. A full system factory reset and reinstall didn't help the issue.

a1pharm, 2015-10-31:
I'm having the exact same problem as Jimmydaux: my Samsung S6 (US Cellular) crashes (the screen goes black and the phone restarts) whenever I try to connect to my 3DR telemetry radio via the Tower App.  I am able to connect for 5-10 seconds before the crash.  I've tried uninstalling/reinstalling the Tower app (and all iterations of it) with the exact same problem occuring every time.  Is there anything 3DR will do to support the Samsung Galaxy S6?

FYI: my Samsung tablet does not have this issue ONLY the S6 does.

Jimmydaux, 2015-11-04:
@ne0fhyk Now that there is at least a second person who admits the same issue, is there anything we can do to get a serious look at this? I am willing to provide any relevant information from my device if you are able to instruct me in what may be helpful. I miss the days of having easy access to tower on my phone.

js349, 2015-11-16:
Is anybody working on this bug? I just received updates to the software and it said some bugs were fixed but not this major bug. Tower connects then reboots the phone. What is the latest on this? Thanks!

a1pharm, 2015-11-19:
It's a shame that the most popular flagship phone simply isn't supported by 3DR.  I think I'm going to have to avoid getting the Solo.

ne0fhyk, 2015-11-19:
@a1pharm It's supported. This issue seems peculiar to certain galaxy s6 (and edge) phones. I have both devices and haven't been able to replicate the issue (unfortunately :( ).

js349, 2015-11-19:
I have the Note 5 and it also has the issue.

ne0fhyk, 2015-11-19:
@Jimmydaux @a1pharm @js349 @Thoin Can you attempt to install and test older versions of [3DR Services](https://github.com/dronekit/dronekit-android/releases) and [Tower](https://github.com/DroidPlanner/Tower/releases) for regression testing.

I'm still unable to replicate the issue, and the logs unfortunately have nothing that looks suspicious.

a1pharm, 2015-11-20:
Yep, I've tried all old version of Tower.  I've tried both Droid Planner and Droid Planner 2 (many versions of each).  I've tried different combinations of 3DR Services and Tower.  Nothing worked - same issue every single time.  Not sure how else to say it: Tower simply doesn't work with the Galaxy S6.  This is the only app I've ever installed on my S6 that just never works.  I would really like to blame my device, but it sure seems like it's a Tower/3DR problem.  Hopefully it will get fixed before the holidays - there will be a lot of Solo returns if Tower doesn't work with a major flagship phone.

Thoin, 2015-11-20:
@a1pharm We all appreciate that your having a problem, we have had problems with this software as well. We're very sorry that it's not working but posts like yours don't help us help you. Your problem does not occur on every Galaxy S6 and we're having problems replication these issues. The developers are doing their best to help and we are doing our best to help them. Let's keep this forum friendly and productive.

@ne0fhyk I will try to test as many combinations that I can tomorrow and let you know of any issues that occur, whether or not I think they're related. Hopefully we can help you nail down this problem.

js349, 2015-11-20:
I think it's the version of Android and not the phone since it does the same thing on the Note 5. My Note 5 has Android 5.1.1.

ne0fhyk, 2015-11-20:
@js349 The devices I am testing with are also running Android 5.1.1, but no luck with reproducing the issue.

lvale, 2015-11-20:
With the last updates to 3DR Services and Tower, both my machines now work without issues

ne0fhyk, 2015-11-20:
@a1pharm this issue only affects vehicles that communicate using a telemetry radio, pointing to an issue with the USB stack in the affected mobile devices.

Solo communicates over Wi-Fi and thus is not subject to the same issue.

As @Thoin pointed out (thanks for comment :) ) complaining about it doesn't help and only distract from the effort. I have a galaxy s6 and s6 edge that don't exhibit the issue, there's no smoking gun in the logs which means, short of getting access to your device for troubleshooting, there's no lead where to start looking to fix the issue.
The best thing you can do to help is to include as many information about your tests to help me nail the issue.

ne0fhyk, 2015-11-20:
@Thoin can you focus the testing on 3dr services, it's likely to be the source of the issue. You only need to change your version of Tower to match the version of 3dr services you're testing.
Thanks!

Jimmydaux, 2015-11-20:
@ne0fhyk I have installed and tested with multiple versions of tower and 3dr services. Nothing that I did helped in any way. 

I had someone mention that they thought this issue had to do the v1 radios and that v2 radios didn't have this issue. I do have v1 radios only so I cannot test this theory. 

I'm not a developer by any means and don't know if it would help at all, but I can provide logs from my crashing device if that would somehow help narrow down the issue. I would definitely need some direction on what to provide and where to retrieve it.

js349, 2015-11-20:
What are the matching versions of Tower and 3DR?

js349, 2015-11-23:
Any more news on this? Have you been able to test different radio versions as Jimmydaux stated?

a1pharm, 2015-12-11:
Hmm, still broken.  I guess the Samsung S6 just isn't a popular enough device to warrant the 3DR developers to create a solution to their problem.  I really *really* want to stay with 3DR, but if they refuse to support my (extremely popular) device, I am going to be forced to jump on the DJI bandwagon...

Jimmydaux, 2015-12-11:
I'm sure that there are at least several people that are having issues that would love to have this working. I am definitely one of those people. 

As has been evidenced earlier in this thread, snide comments wont compel the developers to work on this any faster. Squeaky wheel gets the grease but it seems that there are far squeakier wheels out there. It sounds like there are only a handful of us that are actually having this issue. I am crossing my fingers and remaining hopeful that someone can figure out the issue. 

Let us remain respectful and hopeful that this issue that will be resolved.

hunter3316, 2015-12-11:
@a1pharm : FYI  I'm using a 3rd party 3dr telemetry on my Samsung S6, with no problems.

dubvu2015, 2015-12-31:
I have been having the same issues as @a1pharm and @Jimmydaux since I have owned my Samsung Galaxy S6 Edge (Sprint) since July. My initial thoughts was that the radio was drawing too much current from the phone, causing it to reboot itself. I tested this theory by wiring up the radio to a separate 5v UBEC and another cable for data going to the S6. I am able to use the original Droidplanner, but both Droidplanner 2 and Tower cause the device to reboot. So this does appear to be a software issue found somewhere 5-8 seconds after the connect button has been pressed within the app. I've included a small bit of log data from Tower and 3DR Services from my device. There isn't a lot there because I am unfamiliar with pulling log files, but I isolated a log dump with the keyword "droidplanner". If there is anything I can do to help, please let me know as I want to help fix this problem.
[logcat_12-31-15.txt](https://github.com/DroidPlanner/Tower/files/75481/logcat_12-31-15.txt)

Jimmydaux, 2016-02-07:
I have a new proposition. I will send my phone to one of you devs so you can see and work the issue first hand. Would that help?

TMI2000, 2016-06-06:
I have two samsung devices a Tablet NotePro 12 inch  and a  S5 K-Zoom. When I have created a mission on the Tablet (nice big screen) and I want to upload it to the Solo the Tower app stops within a few seconds. The tablet runs android 5.0.2. The S5 has no problem with connecting and uploading.

Blade77, 2016-06-11:
I am having the same crash issue now on my S5 after updating to the latest tower last week? I have tried reinstalling 3dr services and tower to no avail. I have a Samsung tab 2 and it still works fine.... any solutions out there yet?