DroidPlanner/Tower #1604: Resume control with Solo controller not possible

mtbsteve, 2015-09-19:
Once you put Solo into guided mode by any of the Tower commands the only way to regain control via the Solo controller is to switch to Loiter in Tower. More consistent would be to be able to use the Fly button on the Solo controller to exit any auto fight modes

ne0fhyk, 2015-09-19:
@mtbsteve pressing the `fly` button on the controller always bring you back to `loiter` mode not matter which app changed it.

closed 2015-09-19 by ne0fhyk

mtbsteve, 2015-09-19:
@ne0fhyk sorry no, could you please check again. 
Pressing Fly puts Solo into Land. There is no way to get back frim Guided  into Loiter except setting it in Tower. 
It's 100% reproducible with Solo FW 1.1.5 and Tower beta3.

TShapinsky, 2015-09-19:
@mtbsteve Tower shouldn't be interfering with the communication between the controller and solo. Does it still go into land if you don't have Tower connected?

mtbsteve, 2015-09-19:
Ok, here's a step by step instruction how to reproduce it:
1. power Solo controller and Solo 
2. connect Android device to Solo wifi, launch Tower 
3. wait until Solo got GPS fix
4. In tower, press connect, then arm
5. press takeoff in Tower
6. Solo is in Guided mode, hovering 
7. now press Fly in the Solo controller  shortly- nothing happens. Press Fly for a few seconds- Solo goes into Land mode. 
The only way to get back in Loiter is to set it in the Tower menu.
It's not a showstopper, but annoying and not what you expect.

TShapinsky, 2015-09-19:
@mtbsteve I understand but what I'm trying to clear up is whether it's a problem with tower or with your solo firmware. Because tower isn't responsible for handling the button presses on the controller it is likely that you have a problem with your firmware (probably just a parameter) not with Tower. So if you were to takeoff with tower and then disconnect tower before hitting the fly button would it still go into land. Also, if you setup the A and B buttons in the solo app can you switch without having to use tower or is it permanently stuck in land until you use tower?

mtbsteve, 2015-09-20:
@TShapinsky @Ne0fyhk I can't say if it is a bug in Tower or the Solo firmware. I tested today again and can say that it only happens in Guided mode in Tower. When I fly an Auto mission I can always press Break or hit Fly to bring it back to Loiter. So with all Solo flight modes initiated in the Solo app. 

Today Tower even crashed when I hit the Solo Break button while being in Guided mode. The Solo was fortunately stable hovering  above me until Tower restarted and I could switch to Loiter to regain control. 
This is becoming a safety issue.  Please reopen this and have a look at it. Thanks.

ne0fhyk, 2015-09-23:
@mtbsteve while on the ground, it seems the controller is not able to switch away from any modes when pressing the fly button. However it is able to do so when the vehicle is airborne.

I tried to reproduce your issue but was unable to do so. If I take off in guided using tower, or if I switch to guided mode in the air using tower, I'm always able to go back to loiter by pressing the fly button on the controller.

ne0fhyk, 2015-09-23:
@mtbsteve if it doesn't work in your case, that may signal an issue with your controller.
You can test this by mapping another mode (i.e: fly manual) to the `a` or `b` button, switching to the mapped mode while in flight, and attempt to switch back to loiter using the fly button.

mtbsteve, 2015-10-10:
@fredia Amazingly it works now with Solo firmware 1.2.

>
> @mtbsteve if it doesn't work in your case, that may signal an issue with
your controller.
> You can test this by mapping another mode (i.e: fly manual) to the a or b
button, switching to the mapped mode while in flight, and attempt to switch
back to loiter using the fly button.

ne0fhyk, 2015-10-11:
@mtbsteve great to hear!