DroidPlanner/Tower #1608: Mission: Remaining flight time information missing

ThorstenDP, 2015-09-22:
Hi all,
the latest version is not showing the remaining mission time. I can't find it in the settings but maybe I am missing something?!
Best regards,
Thorsten

squilter, 2015-09-22:
It wasn't very accurate, so I think we just got rid of it.  Now there's a progress bar, which is probably more helpful anyway.

ThorstenDP, 2015-10-04:
Even though it might not be 100% accurate, I think it is important as an additional security feature. Apart from the remaining flight time it would also be good to see the flight time after take off as well. Both are especially important if you are operating at the end of LOS with long flight times where it might take some min until RTL after a battery failsafe.