DroidPlanner/Tower #1609: Altitude during setting up the survey

jostak, 2015-09-23:
Hi, just updated the tower with v3.2.0.beta.3 and found out that after setting the min and max altitude, those limits will be valid only for not survey type mission, means if you select the waypoint the limits will be as you set it, but for survey the limits are 0-200m.

Anyway, this new version gives the flexibility for alt limits and this is great improvement, is it possible to also straight out the alt confusion, means what is the above see level what is above start alt level and such?

I do not have the ability to check it now, but in last version, once I set the guided mode via tablet and changed the alt to 200m (which was the max allowed at that time)  the plane started to descent. The plane was at that time 100m above the lunch alt and the lunch alt was 147 above see level. So it seems to me that the plane wanted to go to 200m above the see level and I hoped that the command will change the alt from lunch.

Thank you for the great work you did with the planner.

ne0fhyk, 2015-09-23:
@jostak thanks for the bug report! The survey issue has been fixed and the next beta (beta 4) should include the fix when it comes out.

Do you mind testing out plane support again once beta 4 comes out, and open another issue if the alt behavior is still inconsistent? The new code should correctly handle the altitude for planes, but I don't have a plane handy myself, and as such I have no way to test asides from using simulations.

closed 2015-09-23 by ne0fhyk

jostak, 2015-09-23:
I will be happy to help whenever I can. After connecting to plane I see 2 altitudes now, one is I believe the  alt of the lunch position and the other is the current plane alt, correct?
Why this alt next to the house icon is changing a lot before arming and right after arming is set to 0?
Since I have 3D locked, my alt above see level shall be arround 147m, but I see from 0 to 50m.

Anyway, even after arming this alt is changing by few m.
During the test gps is solid hdop 1.8.
What are those alts now?

ne0fhyk, 2015-09-23:
@jostak the plane alt was moved to the action bar and it is the one next to the house icon. The plane altitude is now given as AGL (above ground level).

The values beneath the attitude telemetry data are now the vehicle's horizontal speed (hs), and vertical speed (vs). It's normal for these values to fluctuate slightly as they are determined using the gps location, which always fluctuate slightly.

jostak, 2015-09-24:
OK, perfect, so what is the value right next to the AGL alt - on the right, such icon with the arrow up, since it has changed by 1 m, when I put my plane higher by aprox. 1 m, I thought that it's the AGL then.
The ATT telemetry is crystal clear to me.
Later this day if the wind allows, I will test the system.

Thank you so much for your help - I really appreciate the work you did for the model makers like myself.