DroidPlanner/Tower #1618: Error setting max altitude value 3.2.0 beta 4

guiseco, 2015-09-28:
09-28 09:45:14.312  7179  7179 E AndroidRuntime: FATAL EXCEPTION: main

09-28 09:45:14.312  7179  7179 E AndroidRuntime: Process: org.droidplanner.android, PID: 7179

09-28 09:45:14.312  7179  7179 E AndroidRuntime: java.lang.NoSuchMethodError: No virtual method getContext()Landroid/content/Context; in class Lorg/droidplanner/android/fragments/SettingsFragment; or its super classes (declaration of 'org.droidplanner.android.fragments.SettingsFragment' appears in /data/app/org.droidplanner.android-1/base.apk)

09-28 09:45:14.312  7179  7179 E AndroidRuntime: 	at org.droidplanner.android.fragments.SettingsFragment$12.onPreferenceChange(SettingsFragment.java:428)

09-28 09:45:14.312  7179  7179 E AndroidRuntime: 	at android.preference.Preference.callChangeListener(Preference.java:928)

09-28 09:45:14.312  7179  7179 E AndroidRuntime: 	at android.preference.EditTextPreference.onDialogClosed(EditTextPreference.java:145)

09-28 09:45:14.312  7179  7179 E AndroidRuntime: 	at android.preference.DialogPreference.onDismiss(DialogPreference.java:391)

09-28 09:45:14.312  7179  7179 E AndroidRuntime: 	at android.app.Dialog$ListenersHandler.handleMessage(Dialog.java:1257)

09-28 09:45:14.312  7179  7179 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:102)

09-28 09:45:14.312  7179  7179 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:135)

09-28 09:45:14.312  7179  7179 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:5253)

09-28 09:45:14.312  7179  7179 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)

09-28 09:45:14.312  7179  7179 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Method.java:372)

09-28 09:45:14.312  7179  7179 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:899)

09-28 09:45:14.312  7179  7179 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:694)

09-28 09:45:14.322   538   557 W ActivityManager:   Force finishing activity org.droidplanner.android/.activities.SettingsActivity

Linjieqiang, 2015-09-28:
yes,it's the same to me.

ne0fhyk, 2015-09-28:
@guiseco @Linjieqiang beta 5 will be pushed shortly to fix the issue. Thanks for the bug report!

closed 2015-09-28 by ne0fhyk