DroidPlanner/Tower #1619: Error setting map provider preferences 3.2.0 beta 4

guiseco, 2015-09-28:
09-28 09:43:18.542   538   625 I ActivityManager: Displayed org.droidplanner.android/.activities.SettingsActivity: +372ms

09-28 09:43:22.242   538   687 I ActivityManager: START u0 {cmp=org.droidplanner.android/.activities.helpers.MapPreferencesActivity (has extras)} from uid 10180 on display 0

09-28 09:43:22.492  7075  7075 E AndroidRuntime: FATAL EXCEPTION: main

09-28 09:43:22.492  7075  7075 E AndroidRuntime: Process: org.droidplanner.android, PID: 7075

09-28 09:43:22.492  7075  7075 E AndroidRuntime: java.lang.NoSuchMethodError: No virtual method getContext()Landroid/content/Context; in class Lorg/droidplanner/android/maps/providers/google_map/GoogleMapPrefFragment; or its super classes (declaration of 'org.droidplanner.android.maps.providers.google_map.GoogleMapPrefFragment' appears in /data/app/org.droidplanner.android-1/base.apk)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at org.droidplanner.android.maps.providers.google_map.GoogleMapPrefFragment.setupMapboxTileProviderPreferences(GoogleMapPrefFragment.kt:289)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at org.droidplanner.android.maps.providers.google_map.GoogleMapPrefFragment.setupPreferences(GoogleMapPrefFragment.kt:158)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at org.droidplanner.android.maps.providers.google_map.GoogleMapPrefFragment.onCreate(GoogleMapPrefFragment.kt:149)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at android.app.Fragment.performCreate(Fragment.java:2031)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:863)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1067)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at android.app.BackStackRecord.run(BackStackRecord.java:833)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at android.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:1452)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at android.app.Activity.performStart(Activity.java:5990)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2279)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2376)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at android.app.ActivityThread.access$800(ActivityThread.java:147)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1281)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:102)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:135)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:5253)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Method.java:372)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:899)

09-28 09:43:22.492  7075  7075 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:694)

09-28 09:43:22.492   538  1218 W ActivityManager:   Force finishing activity org.droidplanner.android/.activities.helpers.MapPreferencesActivity

09-28 09:43:22.492   538  1218 W ActivityManager:   Force finishing activity org.droidplanner.android/.activities.SettingsActivity

09-28 09:43:22.992   538   619 W ActivityManager: Activity pause timeout for ActivityRecord{3b83f43a u0 org.droidplanner.android/.activities.helpers.MapPreferencesActivity t1743 f}

09-28 09:43:23.852   538  1197 I WindowState: WIN DEATH: Window{b33eae2 u0 org.droidplanner.android/org.droidplanner.android.activities.SettingsActivity}

09-28 09:43:23.852   538   556 I WindowState: WIN DEATH: Window{2956131b u0 org.droidplanner.android/org.droidplanner.android.activities.FlightActivity}

09-28 09:43:23.872   538  1180 I ActivityManager: Process org.droidplanner.android (pid 7075) has died

Linjieqiang, 2015-09-28:
yes,it's the same to me.

ne0fhyk, 2015-09-28:
@guiseco @Linjieqiang beta 5 will be pushed shortly to fix the issue. Thanks for the bug report.

closed 2015-09-28 by ne0fhyk