DroidPlanner/Tower #1622: Manual taking photo or starting video from tower freezes Solo Gimbal (Requires Power cycle of Solo to clear)

bheiland, 2015-10-02:
If I click on camera or video from the tower app, from that point until the next Solo power cycle, the Gimbal becomes frozen and inoperative from the Solo radio control.  No response to attempts to change the pitch angle of the gimbal from the controller.

Must reboot Solo/Gimbal to clear.

bheiland, 2015-10-13:
I realize that it is possible to control the gimbal from tower, but if tower is shut down it would be nice if gimbal control would be returned to the radio controller.

joeynuggetz, 2015-10-18:
Was just going to report this. Looks like touching the video feed window within Tower causes gimbal operations to STOP. I rebooted Solo and controller, connected to wifi, tested gimbals were ok. Then launched tower and connected. Saw video feed. Gimbal still ok. Touched the video screen to make it bigger and gimbal stopped working so there you go. Please test this.

Tower version 3.2.0
Solo firmware 1.2
.

ne0fhyk, 2015-10-18:
@bheiland @joeynuggetz The issue was fixed in commit https://github.com/dronekit/dronekit-android/commit/68470f02bc7dcc23696dfe573281ce73c5d17160.
The fix will be pushed in `3DR Services v1.4.3`.

closed 2015-10-18 by ne0fhyk

joeynuggetz, 2015-10-19:
Someone told me that this was a feature. At least the part where you can control the gimbal in tower now. Didn't realize that. So it sounds like we'll have gimbal control via controller AND tower in services 1.4.3?

ne0fhyk, 2015-10-19:
Yes! Once you exit the screen that lets you take photo/record video, then
control is handed back to the controller.

◤Fredia Huya-Kouadio
    ANDROID TECH LEAD
    3D Robotics
    240.601.6575

joeynuggetz, 2015-10-19:
Excellent. Any eta for a beta we can test?

ne0fhyk, 2015-10-19:
Just pushed `3DR Services v1.4.3 beta 2` which contains the fix! It should be live in the Google Play store in a couple of hours.