DroidPlanner/Tower #1623: support for on-board compass calibrator

rmackay9, 2015-10-03:
Copter-3.4 will include the new on-board compass calibrator from the solo guys.  It will also be in the 3.4.0 plane release which will go out long before the Copter-3.4 release.

Protocol is here:
https://github.com/diydrones/ardupilot/blob/master/libraries/GCS_MAVLink/message_definitions/ardupilotmega.xml#L713

MAVProxy implementation is here: https://github.com/DroneCode/MAVProxy/blob/master/MAVProxy/modules/mavproxy_calibration.py#L115

The new calibration also does scaling (aka soft-iron corrections) which can be very worthwhile on some aircraft.

Here's the equivalent Mission Planner issue: https://github.com/diydrones/MissionPlanner/issues/990

Linjieqiang, 2015-10-04:
when is the copter 3.40 realese?I am looking forward to it.

rmackay9, 2015-10-04:
@Linjieqiang, Copter-3.4 is many months away I'm afraid.  Copter-3.3 is just on it's way out now.

ne0fhyk, 2016-08-08:
Added in pr #1782

closed 2016-08-08 by ne0fhyk