DroidPlanner/Tower #1627: Tower GPS failsafe problem in FollowMe mode

johaninpika, 2015-10-04:
hi guys..

best application on market with a lot of fun.. I would like to use my drone also better with followMe mode...

 ...regarding Failsafe during Follow Me mode when drone loses GPS lock please switch to loiter rather than standard manual. In case we are using only Android app instead paralell RC control it means the Drone will crash when switch to Failsafe.

When using drone with photosesion in sports like cycling, skiing, etc, follow me mode should follow my phone in the pocket so better failsafe let say RTL or Land or Land area waypoint or Loiter everything is better than switch to RC Manual control which is not there...

Please comment or suggest how to...

ne0fhyk, 2015-10-04:
@johaninpika The behavior of the drone during GPS lost failsafe is controlled by its autopilot and not by Tower, so there's nothing we can do in that area. 

Regardless, `loiter`, `rtl` also require `gps lock` to be enabled, hence why the default autopilot behavior in lost GPS failsafe scenario is to switch to a manual mode.

@rmackay9 is there a parameter to change the default behavior on vehicle GPS lost failsafe?

closed 2015-10-04 by ne0fhyk

johaninpika, 2015-10-06:
thanks for explanation.. 

 

but, could it be possible let say somehow manage the way of gps priority between on-board gps and gps from mobile phone:

 

- when on board gps has 6 satelites, use location from on-board GPS

- when on board gps has less satelites, it switch to failsafe but what we can do is to put limit instead 6 to 3 satelites, than i think it will not switch to failesafe and than guide could  be done through tower app

- it would be also nice if there is a possibilitie that a kind of ping delay could guide drone to follow same timedelay

 

- and another feature for Tower app it would be great great support to have manual command under one drop menu for virtual RC commands (up, down, left, right, rotate) and manual trust adjustment to operate drone ... --- all from Tower app, this could help also in case of lost gps lock...

 

Thanks in advance for any additional explanation or help…

 

:j

 

Ta e-pošta je bila pregledana z Avast protivirusnim programom.
https://www.avast.com/antivirus

johaninpika, 2015-10-06:
- and another feature for Tower app it would be great great support to have manual command under one drop menu for virtual RC commands (up, down, left, right, rotate) and manual trust adjustment to operate drone ... --- all from Tower app, this could help also in case of lost gps lock... Thanks in advance for any additional explanation or help… :j

---
Ta e-pošta je bila pregledana z Avast protivirusnim programom.
https://www.avast.com/antivirus

maribeiro, 2015-10-07:
@ne0fhyk  The parameter that defines the behavior of vehicle in GPS failsafe is FS_GPS_ENABLE.
Which can be set to 0 (Disable) or 1 (Land) or 2 (switch to AltHold)