DroidPlanner/Tower #1631: Waypoint reordering buggy

robcee, 2015-10-05:
STR:

1. Enter Flight Editor
2. Draw a sequence of waypoints
3. Insert a new waypoint, change it to a Region of Interest
4. Hold and Drag new waypoint through the list to position 1.

Expected:
ROI gets set to number 1.

Actual:
Bumping the edge of the screen stops reordering waypoint and breaks the original flight plan. Some waypoints become transparent in the list at the bottom and funny interaction ensues.

Alternate fix to drag and drop: Be able to set the waypoint number manually on the waypoint editor itself. Would make the algorithm for reordering waypoints simpler to deal with if it's not being updated dynamically as a result of a faulty drag and drop operation.

robcee, 2015-10-05:
[image]

Screenshot showing mangled flightpath and transparent waypoints in the list area at bottom. Might take a couple of drag attempts for things to get confused. I added an ROI as above then another waypoint with a change of speed order.

ne0fhyk, 2016-08-20:
Fixed by pr #1793

closed 2016-08-20 by ne0fhyk