DroidPlanner/Tower #1632: Tower then 3DR Services apps crashing on radio connect

Thoin, 2015-10-05:
I have 2 different Samsung devices that show this issue. I will first explain the issue, then describe both devices and installed versions. I will attach the appropriate logs to this post as well.

**The Issue**: When I connect the genuine 3DR radio to my devices with the provided USB OTG cable, and click connect, Tower crashes instantly, followed shortly after (less than 5 seconds) by the 3DR Services app crashing.

**Situation**: Pixhawk is powered on and connected to paired radio. I have tested with QGroundControl on my PC and the radios are functioning correctly. Radio on android device has power and is communicating with android as the USB interact lights flash inside the radio and the green light is flashing.

**Devices**: Both devices crash in the very same fashion.
*Device 1*: Samsung Galaxy S5
OS: Stock Android 5.0
3DR Services version 1.4.2
Tower version 3.2.0

*Device 2*: Samsung Galaxy Tab S
OS: Stock Android 5.0.2
3DR Services version 1.4.2
Tower version 3.1.6

Device|Tower Logs|3DR Service Logs
---------|---------------|-----------------------
Galaxy S5|https://drive.google.com/folderview?id=0B-4Pt7stMkc8LUZoZmNhVW9GeXc&usp=sharing|https://drive.google.com/folderview?id=0B-4Pt7stMkc8X0VCcG9rOWp1TlU&usp=sharing
Galaxy Tab S|https://drive.google.com/folderview?id=0B-4Pt7stMkc8bFFZTzZOV0JJakE&usp=sharing|https://drive.google.com/folderview?id=0B-4Pt7stMkc8enduVlhKejg0MFk&usp=sharing

[All Crash Logs for Download in a Single Folder](https://drive.google.com/folderview?id=0B-4Pt7stMkc8ZVBRUnJ3aFBSYUk&usp=sharing)

ne0fhyk, 2015-10-11:
@Thoin this is a duplicate of issue #1590 . Let's continue the conversation there!

Thoin, 2015-10-11:
@ne0fhyk  I would not call these duplicate issues. They are talking about it crashing their phones OS entirely and actually being able to get connected to the autopilot before it crashes. Neither of which occur in my case.

ne0fhyk, 2015-10-11:
@Thoin you're right. The issues in the logs have already been resolved. The updates just haven't been pushed out yet.

From looking at the logs, I'm guessing you're using an autopilot other than Ardupilot, right?

Thoin, 2015-10-11:
Ok. I'll wait for those updates. Thank you.

If you need me to test the updates, let me know. I'm willing and able.

Also, yes. I'm using a pixhawk with PX4 firmware.

ne0fhyk, 2015-10-11:
@Thoin sounds good. I've renamed the issue to reflect the root cause and reopened the issue. 
Support for PX4 is a little bit patchy at the moment but with your feedback, we should be able to fix most of the large issues!

Thoin, 2015-10-12:
@ne0fhyk Alright, count me in. I'll help out however I can.

I'll beta test Tower if you need more people. I know that the PX4 firmware is gaining ground on ardupilot. Let me know how I can help.

A side note, I'm not a skilled programmer but I am an engineer with programming knowledge and willing to learn more.

ne0fhyk, 2015-10-19:
@Thoin I just released `3DR Services v1.4.3 beta 2` to the Google Play store. This release includes the fixes for the bugs you've described here.

As we discussed, since PX4 support is still in progress, solving these bugs will most likely reveal others. 
For any new bugs you encounter, please report them in the [3DR Services issues tracker page](https://github.com/dronekit/dronekit-android/issues) under the [PX4 Support milestone](https://github.com/dronekit/dronekit-android/milestones).

Looking forward to your feedback!

closed 2015-10-19 by ne0fhyk