DroidPlanner/Tower #1635: Missions: Fail in UpLoad and DownLoad mission

AnakinTF, 2015-10-06:
I have same error from upload and download mission, It is not done correctly, don´t write and don´t rear waypoints.

ne0fhyk, 2015-10-06:
@AnakinTF can you provide a better description of the issue.

AnakinTF, 2015-10-07:
Hello, sorry for my poor description.
When I make a flight plan and try to send it through the pixhawk telemetry, it never comes to the FC, I have watched it from the MP not recorded in the FC, also happens when I read the flight plan from tower, I get it, either.

ne0fhyk, 2015-10-07:
Are you connecting your device directly to pixhawk via usb, or are you
using a telemetry radio?

◤Fredia Huya-Kouadio
    ANDROID TECH LEAD
    3D Robotics
    240.601.6575

AnakinTF, 2015-10-07:
I used telemetry radio; today I'll try to do it with cable.

AnakinTF, 2015-10-07:
Hi. 
I think that the error is from mine radio telemetry, now I tryed do with usb port and confección was ok. 
Thank you.

closed 2015-10-11 by AnakinTF