DroidPlanner/Tower #1636: How to setup offline maps?

smocksam, 2015-10-07:
I need more of an explanation as far as how to set this up.

I'm not exactly sure what to paste in for the api section.  

I thought I had it.

It says it's saving maps, but I'm not sure how to test. I guess turning my connection off would probably do it.

The map it shows me is just a road map.  I must have done something wrong since I don't have the satellite maps.

Thanks

busyStone, 2015-10-10:
#1502

ne0fhyk, 2015-10-12:
[image]
[image]
[image]

ne0fhyk, 2015-10-12:
@smocksam the [offline maps wiki](https://github.com/DroidPlanner/Tower/wiki/Offline-Maps) was updated with instructions to retrieve the mapbox id and access token.

closed 2015-10-12 by ne0fhyk