DroidPlanner/Tower #1638: Edit text to speech

CX-Piet, 2015-10-09:
Hi, 
is it possible to edit the text for warnings? Same like in MissionPlanner? 
English terms pronounced in german sounds a bit weird...

THX

ne0fhyk, 2015-10-19:
@CX-Piet the warnings have been externalized to allow [localization](https://developer.android.com/guide/topics/resources/localization.html). All that remains to do is to provide a german version of the warnings.

@squilter @zumbrunnen up for it?

closed 2015-10-19 by ne0fhyk

squilter, 2015-10-19:
Is it just about adding more strings to the german strings.xml?

ne0fhyk, 2015-10-19:
@squilter yes, specifically adding the german equivalent for this section: https://github.com/DroidPlanner/Tower/blob/2e1e5118beddb44e9e7ebf908669e02f3ab94c45/Android/res/values/strings.xml#L529-l571

squilter, 2015-10-19:
Sure, I'll do it later this week.

ne0fhyk, 2015-10-19:
@squilter [sounds good](https://github.com/DroidPlanner/Tower/issues/1657).

zumbrunnen, 2015-10-20:
@squilter if you're using Android Studio: it has a nice Translation Editor. If you need help, lass es mich wissen. :wink:

squilter, 2015-10-20:
@zumbrunnen Thanks for the tip.  The translations editor looks pretty nice!