DroidPlanner/Tower #1640: Tower and antenna tracker.

visper1, 2015-10-11:
Hello, I have a little problem with the Tower application. With my copter works very well. But when I connect APM antenna tracker, so they are Mavlink two devices. Mission Planner with no problem, each device transmits its SYSID and MP asks which want to control. Tower but it can not and displays together two SYSID and making mismatch. I do not need to control the antenna tracker through the Tower, but I need to be tracker SYSID ignored.

rmackay9, 2015-10-11:
I think this is a duplicate of this issue I already raised: https://github.com/DroidPlanner/Tower/issues/1481

visper1, 2015-10-11:
Yes, this is the same problem. Tower have problem with multiple devices on Mavlink. Mission planner have "sysid selector".
[image]

ne0fhyk, 2015-10-11:
@visper1 I'm closing this issue since it's a duplicate of #1481 as @rmackay9 pointed out.
Let's continue the conversation there!

closed 2015-10-11 by ne0fhyk