DroidPlanner/Tower #1663: Phone UI: battery, mode and HDOP values wrap to second line

ggregory8, 2015-10-22:
I'm running a OnePlus One phone with 5.5inch screen. I'm having issue simiar to #602  but logged new issue as this is specific to Phone I believe.
System has normal font size but the battery voltage, mode and HDOP values all wrap around to second line. It seems there is enough space to fit the text but the whitespace border maybe needs to shrink? 
Even if I shrink to smallest font size it still appears to wrap to next line.

See attached screenshot.
[image]

ggregory8, 2015-10-22:
Can we change the location of the warning messages. as currently if we have the drawer out (mostly always in my case) the message is hidden. Could dock the messages at bottom, just above Arm / Dronie buttons? Or alternatively make the warning text smaller and drop the top of the drawer down a little to allow the full message. Should I track this as a separate issue?

ne0fhyk, 2016-04-01:
Fixed by pr #1727

closed 2016-04-01 by ne0fhyk