DroidPlanner/Tower #1668: Sony QX1 & a5000 cameras in survey camera list

SciAero, 2015-10-28:
Can the Sony QX1 & a5000 cameras (same sensor) please be added to the survey camera list.

They both have the same sensor size and lens options as the Nex-5 but a 20MP sensor instead of 16MP.

The sensor resolution is 5456 x 3632.

ne0fhyk, 2015-10-29:
@SciAero here's the camera info for the Nex-5. Can you update the relevant parts for the Sony QX1 and a5000:

```
<?xml version="1.0" encoding="utf-8"?>
<cameraInfo>
    <Name>Sony Nex5 20mm</Name>
    <SensorWidth>23.4</SensorWidth>
    <SensorHeight>15.6</SensorHeight>
    <SensorResolution>14.2</SensorResolution>
    <FocalLength>20.0</FocalLength>    
</cameraInfo>
```

SciAero, 2015-10-29:
Thanks Fredia,
Here are the update version for QX1 with 16mm and 20mm lens respectively:

<?xml version="1.0" encoding="utf-8"?>
<cameraInfo>
    <Name>Sony QX1 16mm</Name>
    <SensorWidth>23.4</SensorWidth>
    <SensorHeight>15.6</SensorHeight>
    <SensorResolution>20.5</SensorResolution>
    <FocalLength>16.0</FocalLength>    
</cameraInfo>

<?xml version="1.0" encoding="utf-8"?>
<cameraInfo>
    <Name>Sony QX1 20mm</Name>
    <SensorWidth>23.4</SensorWidth>
    <SensorHeight>15.6</SensorHeight>
    <SensorResolution>20.5</SensorResolution>
    <FocalLength>20.0</FocalLength>    
</cameraInfo>

SciAero, 2015-10-29:
Thanks Fredia,

Here are the update version for QX1 with 16mm and 20mm lens respectively:

<?xml version="1.0" encoding="utf-8"?>
<cameraInfo>
    <Name>Sony QX1 16mm</Name>
    <SensorWidth>23.4</SensorWidth>
    <SensorHeight>15.6</SensorHeight>
    <SensorResolution>20.5</SensorResolution>
    <FocalLength>16.0</FocalLength>
</cameraInfo>

<?xml version="1.0" encoding="utf-8"?>
<cameraInfo>
    <Name>Sony QX1 20mm</Name>
    <SensorWidth>23.4</SensorWidth>
    <SensorHeight>15.6</SensorHeight>
    <SensorResolution>20.5</SensorResolution>
    <FocalLength>20.0</FocalLength>
</cameraInfo>

-- 
*Joshua Portlock*
*Director - Chief Technology Officer*
10/8 Booth Place, Balcatta, WA, 6021, Australia
Direct: +61 (0) 8 6231 0270
Mobile: +61 (0) 409 011 447
*www.Sci.Aero <http://www.sci.aero/>*
[image: Inline image 4]  [image: Inline image 1]

This message contains confidential information and is intended only for the
individual named. If you are not the named addressee you should not
disseminate, distribute or copy this e-mail. Please notify the sender
immediately by e-mail if you have received this e-mail by mistake and
delete this e-mail from your system. E-mail transmission cannot be
guaranteed to be secure or error-free as information could be intercepted,
corrupted, lost, destroyed, arrive late or incomplete, or contain viruses.
The sender therefore does not accept liability for any errors or omissions
in the contents of this message, which arise as a result of e-mail
transmission.

ne0fhyk, 2015-10-29:
@SciAero thanks for the update. The camera info was added by PR https://github.com/dronekit/dronekit-android/pull/322

closed 2015-11-03 by ne0fhyk