DroidPlanner/Tower #1672: Editor does not show waypoints containing a 0,0,0 reset ROI with lastest 3DR Services (1.4.3)

billfor, 2015-11-05:
With the latest release of 3DR services (1.4.3) they added a resetROI mission item (https://github.com/dronekit/dronekit-android/issues/312).

If an old style mission is loaded into the drone that resets the ROI by setting lat,long,alt to 0, then tower will scroll to the centerpoint of the mission, but no waypoints will be shown (mission appears empty).

Downgrading to 3DR services 1.4.2 fixes the problem. 

The drone must be connected to show this problem.
To reproduce:

1. From PC , upload a flightplan to the drone via missionplanner  that contains an ROI with lat,long,alt set to 0.
2. From tablet,  sideload 3DR services 1.4.2 (prior version)
3. From tablet, connect to drone via tower and download the flightplan uploaded in step 1 from the drone. Verify that you can see/edit the flightplan.
4. Save the flightplan locally from tower to the tablet and disconnect. 
5. Install latest 3dr services 1.4.3
6. Restart tower. DO NOT CONNECT THE DRONE. 
7. Load the locally saved flightplan saved in step 4, verify that you can see edit it in the editor.
8. Connect to the drone in tower.

Now the flightplan will disappear and show no waypoints.   Even if you import it again from the local tablet file, the flightplan will briefly appear and erase. In order to get it back, you have to disconnect from the drone, but the moment you reconnect to the drone the plan will disappear again.

I would attach a sample flightplan I used to repro but I don't have write access.

ne0fhyk, 2015-11-05:
@billfor which version of Tower are you running?

billfor, 2015-11-05:
This is for the 3.2 (current release) version. Didn't see any fixes in the beta changelog. I'll try 3.2.1. Thanks!

ne0fhyk, 2015-11-05:
@billfor Actually i made a mistake. It's part of the second beta release for version 3.2.1, but it's not out yet. Only the first beta is out.
It was fixed by pr #1267

billfor, 2015-11-06:
I pulled and built the 3.2.1.debug.2 confirmed this issue is fixed currently. Flightplan displays and a new gear icon indicates reset ROI. Thanks!

closed 2015-11-06 by billfor

ne0fhyk, 2015-11-06:
@billfor (incoming) Version 1.4.4 for 3DR Services will also fix the issue for current and older versions of Tower.