DroidPlanner/Tower #1684: How to set GimbalApi

tongzahub, 2015-12-06:
how to  set GimbalApi on button OnClick 

GimbalApi.getApi(dpApp.getDrone()).updateGimbalOrientation(90,90,90,null);
GimbalApi.getApi(dpApp.getDrone()).startGimbalControl(null);

TShapinsky, 2015-12-06:
the listener passed to `startGimbalControl` must be non-null. Also you may want to store the GimbalApi in a variable so that you don't have to get a new one everytime you make an Api call.

tongzahub, 2015-12-07:
could you show me some example for this case? thank you

chaviw, 2015-12-07:
You can take a look at this class
https://github.com/DroidPlanner/Tower/blob/b47ef5305e5e91f9291bdb347a0462fc1524d49d/Android/src/org/droidplanner/android/fragments/widget/video/FullWidgetSoloLinkVideo.kt

It's in Kotlin but you should be able to figure out what it's doing. Essentially, the gimbal is manually controlled when the user touches the video stream view

closed 2015-12-07 by ne0fhyk