DroidPlanner/Tower #1691: Unable to set "text to speech" language

wcfung, 2015-12-29:
After  Android was upgraded to V5.1.1 in my Samsung tablet, both Tower and Droidplanner 2 speech output do not work.  Before upgrading , all the announcement and warning worked perfectly. 

The error message upon opening the app is "Unable to set text to speech language".

I tried to uninstall and re-install both Tower and Droidplanner 2.  It did not help.

Pls help.

ne0fhyk, 2016-08-10:
@wcfung, I'm not able to replicate the issue.
In the android settings app, can you go to 'Languages & input' -> 'Text to speech output' and make sure that you have the correct language data.

wcfung, 2016-08-12:
I am not able to get it work even though I went to "text and language
output" to set the correct language data.
The fact is : the speech output suddenly did not work without me changing
any setting. The only thing I am not sure is : the Android version was
automatically updated in my Samsung tablet , and IT SEEMS (again , not
sure) since then, it stopped working.
Any other clue you may suggest ?
Thank you.

ne0fhyk, 2016-08-13:
@wcfung I remember the Samsung tts giving us some trouble in the past. Can you test with the Google tts, and see if that solves things: https://play.google.com/store/apps/details?id=com.google.android.tts

wcfung, 2016-08-17:
The tts now functions with the Google tts app. Thanks

ne0fhyk, 2016-08-18:
Great to hear!

closed 2016-08-18 by ne0fhyk