DroidPlanner/Tower #1694: event Arm ,but don't need set HomeDrone

tongzahub, 2016-01-08:
event Arm in Utils.java 

I need Arm ,But don't need set HomeDrone 
how to  solution

code in Utils.java 
public static void getArmingConfirmation(FragmentManager fm, final Drone drone){

		getActionConfirmation("arm", fm, new Runnable() {
			@Override
			public void run() {

				VehicleApi.getApi(drone).arm(true);

			}
		});
	}
[image]

tongzahub, 2016-01-08:
help me please!...

ne0fhyk, 2016-01-08:
@tongzahub this is not the place for these type of questions. Can you instead ping the [dronekit-android gitter channel](https://gitter.im/dronekit/dronekit-android).
Once you do can you clearly explain what you're doing now, and its results and what you expect the results to be instead.

closed 2016-01-08 by ne0fhyk