DroidPlanner/Tower #1695: Pause button causes Copter-3.3.2 to fly off

rmackay9, 2016-01-09:
I've reproduced this problem just now after being reported by Stephan on the Copter-3.3-beta testing thread.

I'm looking into the cause as a top priority but it's not clear to me yet if the problem is in Tower or in Copter.

rmackay9, 2016-01-11:
I think there may be an issue with Tower or 3DR Services caching an old version of the last known vehicle position.  This cache may persist beyond a single flight and may somehow be being used the next time the Pause button is pushed.

rmackay9, 2016-01-11:
The report can be found in the Copter-3.3-beta testing thread (search for "Stephan"):
http://diydrones.com/forum/topics/copter-3-3-beta-testing?xg_source=activity&id=705844%3ATopic%3A1964465&page=425#comments

here's the dataflash log (which doesn't show much): http://diydrones.com/forum/attachment/download?id=705844%3AUploadedFile%3A2163944
here's the video: http://youtu.be/WGdAYg2UGJU

rmackay9, 2016-01-11:
I believe this will recreate the issue although I have not tested on a real vehicle because I destroyed my IRIS today:
1. start up the vehicle in location A, connect with Tower
2. unplug vehicle and move to location B (leave Tower on and do not push disconnect)
3. plug in vehicle, Tower will reconnect automatically but may not request data stream to be sent at regular intervals (?) meaning that position is not updated
4. take-off and switch to follow-me
5. push Pause button, vehicle will attempt to fly to Location A

ne0fhyk, 2016-01-11:
@rmackay9 The `pause` button sends a guided point to the vehicle's current location, causing the vehicle to stop in place if its reported GPS location matches its actual location.
I took a look at the code, and realize we're not doing checks on the validity of the reported GPS location.
That could be the cause of the issue.

ne0fhyk, 2016-01-11:
@rmackay9 Although, on the posted video, the user was doing `follow-me` before pressing `pause`, which would imply its vehicle had GPS lock. 
Do you have access to the tlog data. From it we should be able to see which guided point coordinates were sent to the vehicle when it pressed `pause`.

rmackay9, 2016-01-12:
Follow-me actually doesn't require that the tablet know the vehicle location and it's the out of date vehicle location within the tablet that is the problem.

mtbsteve, 2016-01-13:
Thanks guys for investigating the issue.
Please find the tlog file of the day attached.
The error was reproducible and never occurred with the 3.2 release nor with the current Solo FW in combination with Tower. 

[log_usb_2016_01_03_11_53_20.zip](https://github.com/DroidPlanner/Tower/files/89521/log_usb_2016_01_03_11_53_20.zip)

ne0fhyk, 2016-01-13:
@mtbsteve thanks for the logs!

@rmackay9 I'm going to review the logs for additional data. In the meantime, we're planning to update the logic for the `pause` button on solo vehicles and vehicles running AC 3.3  by using `BRAKE` instead as discussed in the dev's call.

Since the issue doesn't seem to affect AC 3.2, for now we can leave the current implementation as is for these versions. We'll still add some checks to validate the gps position value before usage.

rmackay9, 2016-01-14:
@ne0fhyk, great, thanks a lot!

ne0fhyk, 2016-05-16:
Fixed with the release of Tower v3.2.1!

closed 2016-05-16 by ne0fhyk

rmackay9, 2016-05-16:
great!