DroidPlanner/Tower #1696: Random GPS positioning

Wolf68k, 2016-01-13:
I have a Nexus 7 2013 (Android 6.x), also tested on a Nexus 7 2012 (Android 4.4.2), when getting the location of the tablet in Tower, also tested with DroidPlanner2, the location is fine at first (about 2mins) then it randomly shoots to a completely different location. While the new location is near where I actually am it's still about 30ft away. At times it seems to picks about 4 different locations, the 4th being the rarest which is my actual location. It's because of this that I have yet to try the follow me mode for obvious reasons; I don't need it shooting off to some place I'm not.

Not that it should matter but just in case it does; I'm flying an IRIS+

ne0fhyk, 2016-01-13:
@Wolf68k can you make sure your gps settings are set to high accuracy. If they are correctly set, then it might be an issue with the device gps. 
You can confirm by opening another 'map' app, and observing the `my location` marker behavior.

Wolf68k, 2016-01-14:
High accuracy is seems is the default setting because that's what it was set to when I finally found it.
As for the other map app testing.
So far mostly just use the Google Map app and Map Quest because of that satellite view so I have a better frame of reference of where the blue dot is. It's still kind of random, more so with Google Maps.
I've also been using GPS Test and taking the cords from that and applying them to the Google Maps website on my desktop, but I don't think it's updating nearly as much.

I noticed this. At the time I made the report it was all over the place and as I said about 100ft away. Later that evening after giving it about 5-10mins it seems to lock on to me more. Then that night it was close but still not as accurate as my iPhone.

Let me watch this for a couple more days, at various times of day, and I'll get back do you.