DroidPlanner/Tower #1704: Survey does not work!

rsilmakuvaus, 2016-02-18:
Hey, I`m new here so I hope this comes to right place. Survey doesn`t draw aeria. Distance flight time  etc. shows 0,0. I have tried several times to reload app and trie diffrent phones and its always the same. Does anyone else have same broblem?

CreativeBR, 2016-02-18:
It is happening here too. The flight altitude of the survey properties is red.

wolkstein, 2016-02-21:
Same here,
flight altitude is red.
But If Tower is conected to an vihicle survey function works normal. So there is no way to create an survey offline.

ne0fhyk, 2016-02-22:
@rsilmakuvaus Thanks for the bug report, I'll take a look at the issue.

ne0fhyk, 2016-02-22:
The issue has been fixed and the fix will be released with the 3DR Services v1.5.2 release.
Thanks!

closed 2016-02-22 by ne0fhyk

ne0fhyk, 2016-02-23:
As a workaround, you can downgrade to the [3DR Services v1.5.0](https://github.com/dronekit/dronekit-android/releases/tag/3DRServices-v1.5.0.4) to recover the functionality.

rsilmakuvaus, 2016-02-24:
Thousand thanks :)
Best regards hannu