DroidPlanner/Tower #1707: Error creating Survey/Spline Survey

franknitty69, 2016-02-22:
In 3.2.0 and or 3.2.1 beta 1, Survey or Spline Survey missions are not being created correctly.

**Steps to reproduce**
-open tower
-navigate to the editor
-select the spline wayment menu
-set the waypoint type to survey
-draw a survey

**results**
The Distance, Flight Time, Area, Grid Length, Pictures and No of Strips report 0 and there isn't a grid on the map. Also altitude is highlighted red.

**expected results**
Survey is created correctly

ne0fhyk, 2016-02-22:
@franknitty69 thanks for the bug report. This is caused by a bug with 3DR Services which has been fixed.
The fix will be included in the 3DR Services v1.5.2 release.

closed 2016-02-22 by ne0fhyk

craigo-, 2016-02-22:
@ne0fhyk, can you please point me to a 3DR Services binary, version 1.5.2 or the last known working version? I have a bunch of survey missions to fly tomorrow and am currently dead in the water...