DroidPlanner/Tower #1714: Solo update 2.1.0 and tower survey

rsilmakuvaus, 2016-03-10:
Hey. Does anyone else have broblem with camera trigger on tower on survey mission. It wont take pics. Broblem started yesterday when I updated Solo 2.1.0.

Best regards Hannu

JamesMasterman, 2016-03-18:
Yes, I have the same problem. Distance trigger commands in missions no longer work. This all worked fine in any version prior to 2.1.0. It's a big issue for me as I use my solo for Survey. Any work around?

rsilmakuvaus, 2016-03-18:
There was missing a line in shot manager. If someone has way to fix the problem, please let us know :)

http://www.3drpilots.com/threads/how-to-do-camera-trigger-in-tower.4788/

jmachuca77, 2016-03-18:
As a work around, you can still use the time-lapse function on the GoPro to take pictures every X seconds. The only drawback is you will have to georeference by time offset instead of using the CamTrigger messages.

jmachuca77, 2016-03-18:
I do think however that this is not an issue with Tower, this is probably an issue with the Solo firmware.

ne0fhyk, 2016-03-18:
@jmachuca77 @rsilmakuvaus can you confirm no pictures are taken during the whole length of a survey mission? 
The survey logic was recently updated to only start taking pictures after the first survey waypoint has been reached. Could it be that you're expecting pictures to be taken between takeoff and the first survey waypoint, and cancelling the mission when that doesn't occur?

rsilmakuvaus, 2016-03-18:
That time lapse is good news. I`ll test it tomorrow. Geotagging is my biggest worry in time lapse, but tomorrow I know more.
And yes I confirm, no pictures were taken during whole mission.

rsilmakuvaus, 2016-03-19:
Time lapse works well. Georef don't. Time-lapse had to start manually. Tower app always throw to single photo. But I managed to get the photos and I geotagged them with with mission planner estimate offset. I deleted most of the pics (before and after the mission) Pre process OK > location KML Not ok, Geotag images OK. When I opened pics in google or Pix4D all pictures were in one location. So mapping can't be done. When I open pictures location file, all pictures have different location. But when I open file in google or Pix4D all pics are in one location (no cam trigger message) > map goes wrong.
 So i tried take only one georef image that i could get just map with location> map still goes wrong??. After i took off every georef image map is ok. So help me, what I do wrong. 
What I need to do is to locate every pic to KML and create a map with Pix4D.
I know that this isn't right forum but I'm not very smart with these and you guys are the best so you might tell where to find help :)

ne0fhyk, 2016-03-19:
@jmachuca77 @rsilmakuvaus the source of the issue was found, and it'll be fixed with the solo firmware release 2.3.0.
In the meantime, if you're familiar with the code on the companion computer, you might be able to add a workaround to restore the functionality.

SPerkele, 2016-03-19:
@ne0fhyk Great news!  I'm listening, if you will share how to do the workaround.  :)

jmachuca77, 2016-03-19:
@ne0fhyk I guess you found the issue, I read your comment too late. But yes, no pics during the whole survey, the mission is created correctly by tower, and something happens with Mission planner, so thats why I said thought the issue was on Solo. But you found that :)

JamesMasterman, 2016-03-19:
@ne0fhyk that's great news, do you have a link to the workaround before 2.3.0 comes out? I'm comfortable with modifying scripts on the Solo. Is it what @rsilmakuvaus linked to?

Another thing I noticed on 2.1.0 is a whole lot of mavlink packets with CRC errors. If you run with a debugger you can see the parser class detects and ignores them at the first CRC byte check, but there are so many. I do not see this on any other solo release or SITL. I'm not sure they are even mavlink, just seem to be junk. The message ids are reported to be 22,1,154 mainly, but there are sometimes others.

SPerkele, 2016-03-22:
If someone managed to restore the trigger functionality. Please let me know.  I would be happy if can try this out before 2.3.

bheiland, 2016-03-24:
Is there a way to revert to a previous Solo version or firmware?

jmachuca77, 2018-08-16:
This was a Solo firmware issue that has since been resolved.