DroidPlanner/Tower #1720: Tower app crashes on Android 4.0.4 when no wifi

mariansoban, 2016-03-23:
Tower application always crashes while start when there is NO ACTIVE WIFI connection
LOG:
[crash.log.txt](https://github.com/DroidPlanner/Tower/files/186018/crash.log.txt)
Fix: 
https://github.com/DroidPlanner/Tower/blob/develop/Android/src/org/droidplanner/android/utils/NetworkUtils.java#L53
    public static String getCurrentWifiLink(Context context) {
...
        final String connectedSSID = connectedWifi == null ? null : connectedWifi.getSSID().replace("\"", "");
->
        final String connectedSSID = connectedWifi != null && connectedWifi.getSSID() != null ?  connectedWifi.getSSID().replace("\"", "") : null;

Environment: Lenovo A660 with Android 4.0.4

emerich68, 2016-04-01:
have the same problem in galaxy tab 10" & android 4.0.4.
Can you send me a compiled version with your correction (.apk)

mariansoban, 2016-04-01:
Here you are. Please note, that the version contains my API key for google maps and is not official release.
[tower-release.3.2.0-sobi-3-bugfix_wifi.zip](https://github.com/DroidPlanner/Tower/files/199274/tower-release.3.2.0-sobi-3-bugfix_wifi.zip)

emerich68, 2016-04-01:
it works , I can not setup mapbox id , but it works,
thanks a lot
we hope to be included in the official version

ne0fhyk, 2016-04-01:
@mariansoban thanks for the bug report and fix! The fix will be included in the next beta release.

closed 2016-04-01 by ne0fhyk