DroidPlanner/Tower #1730: Add wiki for setup of Uvc video stream

guiseco, 2016-04-05:
[image]
[image]
[image]
[image]
[image]

guiseco, 2016-05-03:
[image]

ne0fhyk, 2016-05-16:
Successfully completed by @guiseco!

closed 2016-05-16 by ne0fhyk

kikislater, 2016-07-13:
Is it possible to add chipset compatibility to the wiki ? I think chipset is more important than a link to aliexpress ...
Which chipset is onboard with the uvc white you reference on the wiki ? Does UTV007 will work ?