DroidPlanner/Tower #1733: Assigning value to SYSID_THISMAV other than 1 causes 3DR Service stoppage in Tower when connecting

BigOkie, 2016-04-13:
To recreate, change the SYSID_THISMAV value to something other than 1 via MP or APM Planner (in my case I changed it to 128 since it was halfway inbetween the upper and lower range).  Opening Tower on the Android, then trying to connect would return a response about the 3DR Services stopping.  When I changed the SYSID_THISMAV value back to 1 it worked ok again.

tribbloid, 2016-08-28:
+1, though I doubt if any android device have enough USB for that many downlinks. (non p2p WiFi? Never heard of it)

ne0fhyk, 2016-09-25:
Closing at I'm not able to reproduce the issue with the latest beta for Tower.

closed 2016-09-25 by ne0fhyk

tribbloid, 2016-09-27:
Duh, you are not even using 3DR service now :)