DroidPlanner/Tower #1740: Tower app freeses regularly on Samsung Galaxy S5 phone

clovett, 2016-04-21:
I'm using Tower app to do Spline Survey mission.  The mission works great, but after completing the mission I'm finding the Tower app often freezes a this point.  Restarting the app doesn't help.  I usually have to reboot the phone to get it working again.

ne0fhyk, 2016-05-11:
@clovett I have not been able to reproduce the issue on the latest beta (beta 6) for Tower. Can you test that version and let me know if you still experience the same issue: https://play.google.com/store/apps/details?id=org.droidplanner.android.beta

ne0fhyk, 2016-05-22:
@clovett I'm closing this issue for lack of additional information. Feel free to reopen it if the issue is still present on the latest version.

closed 2016-05-22 by ne0fhyk