DroidPlanner/Tower #1747: UVC widget doesn't show the video

fliegermati, 2016-05-12:
Hi,

i bought a uvc device with the arkmicro chip Form aliexpress. 
But when i connect it with a USB host Adapter  to the tablet i only see this:

[image]

[image]

I tried to power the USB host with a external power supply, without ans effect.
Tower Version: 3.2.1 beta 7

Is the UVC device broken?

Can anybody help me?

Fliegermati

guiseco, 2016-05-13:
Do you have another tablet or cell phone to test ? Try connecting only UVC to your tablet.

fliegermati, 2016-05-16:
Okay, without the usb host the uvc device works in Tower now! So i should connect the 433mhz groundstation with bluetooth to the tablet

guiseco, 2016-05-16:
.. or try another USB hub -> http://www.ebay.com/itm/141564311626

ne0fhyk, 2016-05-18:
@fliegermati I'm closing this issue, since it seems to be resolved.

closed 2016-05-18 by ne0fhyk