DroidPlanner/Tower #1752: Can't access tlogs and flight history

TheAeroScout, 2016-05-22:
Guys I've been upgraded to Tower 3.2.1 but I'm now having troubles finding the telemetry logs. There is no longer a 'flight history' tab and when I search there's no tlogs. Has this feature been taken away or am I just looking in the wrong location? Cheers.

guyppp, 2016-05-23:
I also had this issue, but searching through the Device Storage on my Samsung S5, I found a collection of Tlogs here: Local Storage > Device Storage > Android > data > org.droidplanner.services.android > files> tlogs

TheAeroScout, 2016-05-24:
Thanks, I'll check it out to make sure that's where they live.
It would be great to have the 'flight history' tab back (if possible) please, I really found it useful. An extra added feature would be to be able to export .tlogs from that screen, perhaps via email. Would make it nice and easy to get .tlogs from Android across to Mission Planner without having to plug in.
Cheers

TheAeroScout, 2016-05-25:
I've found that if I turn off the Android device and then on again, the latest tlog appears. If I don't do the restart then it's not listed. Pleased that I can find them again, thanks for your help. :)

ne0fhyk, 2016-08-08:
Flight history has been added through pr #1779, and will be included with the next Tower release.

closed 2016-08-08 by ne0fhyk

FLYBYME, 2016-08-16:
Well I really needed this the other day. Came to use it and it was gone. Why was it removed?

ne0fhyk, 2016-08-16:
@FLYBYME UI and functionality revamp