DroidPlanner/Tower #1756: How to add a camera to the application

jhonyagroscan, 2016-05-27:
Greetings, I want to add a multispectral camera to the application to work with a 3dr Only someone could help me as I do that

guyppp, 2016-05-29:
There's a good guide here http://www.mapir.camera/blogs/guide/48987139-adding-the-mapir-camera-to-mission-planner-and-tower 
It's specific for mapir cameras, but if you know the parameters for your camera it should work

kikislater, 2016-05-30:
I think there is a problem in his xml.
Under Mission planner : 
Survey2 + 3.97mm Lens (Standard Lens Included):

```
<Camera>
<name>Survey2 3.97mm 16MP</name>
<flen>3.97</flen>
<imgh>4608</imgh>
<imgw>3456</imgw>
<senh>6.17472</senh>
<senw>4.63104</senw>
</Camera>
```

Under Tower : 
Survey2:

```
<cameraInfo>
<Name>Survey2 16MP</Name>
<SensorWidth>4.608</SensorWidth>
<SensorHeight>3.456</SensorHeight>
<SensorResolution>16</SensorResolution>
<FocalLength>3.97</FocalLength>
</cameraInfo>
```

Sensor width and height does not have the same size in Mission planner and Tower for the same camera ! It's a non sense. It should be this in Tower : 
```
<cameraInfo>
<Name>Survey2 16MP</Name>
<SensorWidth>4.63104</SensorWidth>
<SensorHeight>6.17472</SensorHeight>
<SensorResolution>16</SensorResolution>
<FocalLength>3.97</FocalLength>
</cameraInfo>
```

guyppp, 2016-05-31:
well spotted...I've emailed the people at Mapir to confirm. Will post their reply here when it comes. Thanks for pointing it out! I am also uncertain why the height is greater than the width, as in normal camera orientation (i.e. with the shutter button at the top) the width of the image & sensor must be greater than the height??

kikislater, 2016-05-31:
I tell him on mapir facebook, his answer bellow : 

> MAPIR Camera Looks like one of the sensor size numbers are swapped. Width is always larger than height. We'll get it swapped on the site shortly. Thanks

jhonyagroscan, 2016-05-31:
did the procedure and I could add my application multispectral camera to my problem was put in the name of xml with special caracterees rename and I worked perfectly here xml code
<cameraInfo>
  <Name>RedEdge MicaSense</Name>
  <SensorWidth>4.8</SensorWidth>
  <SensorHeight>3.6</SensorHeight>
  <SensorResolution>1.3</SensorResolution>
  <FocalLength>5.5</FocalLength>
</cameraInfo>