DroidPlanner/Tower #1757: Crash on Connect (using USB Telemetry)

flo-vienna, 2016-06-05:
Hi

I recently started having problems using the usb telemetry module with droidplanner / droidplanner2
(Right now I prefer those over the new Tower software).

Steps to reproduce:

I plug-in the USB telemetry module, it lights up, asks me what software I want to use, I select either droidplanner and click on "Just once". The software opens but as soon as I click "Connect" the software crashes. See the log entries below.

Tablet: Vido-M82 Pro (U357P_v2_0_wxga_yd) using Android 5.1 (Kernel 3.10.65+ )

Software versions: (both are affected in the same way)
Droidplanner 1.2.0
Droidplanner2 2.8.6_RC3 

Error logs:

06-05 10:51:56.584 E/DirectVolume(224): 'usbotg' is pulled out.!!
06-05 10:52:03.330 D/UsbHostManager(777): Added device UsbDevice[mName=/dev/bus/usb/001/006,mVendorId=4292,mProductId=60000,mClass=0,mSubclass=0,mProtocol=0,mManufacturerName=Silicon Labs,mProductName=CP2102 USB to UART Bridge Controller,mSerialNumber=0001,mConfigurations=[
06-05 10:52:03.330 D/UsbHostManager(777): UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=50,mInterfaces=[
06-05 10:52:03.330 D/UsbHostManager(777): UsbInterface[mId=0,mAlternateSetting=0,mName=CP2102 USB to UART Bridge Controller,mClass=255,mSubclass=0,mProtocol=0,mEndpoints=[
06-05 10:52:03.330 D/UsbHostManager(777): UsbEndpoint[mAddress=129,mAttributes=2,mMaxPacketSize=64,mInterval=0]
06-05 10:52:03.330 D/UsbHostManager(777): UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=64,mInterval=0]]]]
06-05 10:52:03.377 W/IntentResolver(777): resolveIntent failed: found match, but none with CATEGORY_DEFAULT

Log output using catlog for the droid planner app after clicking the "Connect" button :

java.lang.ArrayIndexOutOfBoundsException: length=0; index=-1
	at android.hardware.usb.UsbDevice.getInterface(UsbDevice.java:232)
	at com.hoho.android.usbserial.driver.Cp2102SerialDriver.open(Cp2102SerialDriver.java:82)
	at org.droidplanner.android.communication.connection.usb.UsbCDCConnection.openUsbConnection(UsbCDCConnection.java:35)
	at org.droidplanner.android.communication.connection.usb.UsbConnection.openAndroidConnection(UsbConnection.java:85)
	at org.droidplanner.android.communication.connection.AndroidMavLinkConnection.openConnection(AndroidMavLinkConnection.java:57)
	at org.droidplanner.core.MAVLink.connection.MavLinkConnection$1.run(MavLinkConnection.java:79)
	at java.lang.Thread.run(Thread.java:818)

br 

Florian

ne0fhyk, 2016-06-09:
@flo-vienna thanks for the bug report, but these versions are no longer supported.
Feel free to reopen the issue if you encounter it on Tower as well.

closed 2016-06-09 by ne0fhyk

flo-vienna, 2016-06-09:
Hi

I fully understand your point and will test it soon with Tower.
Unfortunately, Tower does not offer the same functionality as those older programs did so that was the reason I was trying to get the older ones to work.

ne0fhyk, 2016-06-09:
Thanks for the feedback. What functionality do you think is missing in
Tower compared to the older versions?

flo-vienna, 2016-06-11:
Hi 

Tried it with Tower now, connected the telemetry module via usb, started tower, click connect: Tower asks me if I want to allow 3DR Services to access the USB device. I click Ok.
Then in Tower I click "Connect" and I get the following error message: "No usb devices to access"

In catlog I get: 
UsbCDCCConnection Error setting up device: No usb interfaces to access.
UsbCDCCConnection java.io.IOException: No usb interfaces to access.  
UsbCDCCConnection com.hoho.android.usbserial.driver.Cp2102SerialDriver.open(Cp2102SerialDriver.java:85)
UsbCDCCConnection org.droidplanner.services.android.communication.connection.usb.UsbCDCConnection.openUsbDevice(UsbCDCConnection.java:145)

Now I managed to get it to work once with Tower like this:
I connected the module, told android to open it with 3DR Services, then started tower and clicked on connect.
It seems that my tablet / android / tower is for some reasons picky on wether I unplug the OTG Hub itself of the device connected to the hub.
If I unplug the OTG hub itself and re-plug it in it normally does NOT work. If I plug the OTG hub in, try to connect, then close the app, unplug the device only, replug it and then it normally works.
(This works for droidplanner and tower ).