DroidPlanner/Tower #1760: Is it support if it is connected via Easycap device ?

kushanshah11, 2016-06-16:


Genestealer, 2016-07-10:
I think you mean is the stk1160 EasyCap type android USB capture cards supported for video streaming, in a similar way to the UVC video stream?

guiseco, 2016-07-20:
you can use a app called EasyCap Viewer and streaming video to Tower.

ne0fhyk, 2016-07-20:
@kushanshah11 I'm closing this issue since you got the answer from @guiseco!

closed 2016-07-20 by ne0fhyk

kushanshah11, 2016-07-21:
@guiseco  i don't want to use EasyCap Viewer ,
i just want to know that Tower is supported stk1160 or utv007 ?

kikislater, 2016-07-21:
It's usefull information. I don't understand why nobody want to answer in other thread also ! 
In my case I bought xyusb like recommandation. May be they don't test with these devices but it is sad they don't want to tell or share. For an open source software it is not common ...

ne0fhyk, 2016-07-21:
@kikislater This community shares freely when it comes to information, so if no one is answering, it's more likely that no one has experiences with this specific hardware.
In that case, you can contribute as well by testing, and reporting your findings through edit to the [Tower wiki](https://github.com/DroidPlanner/Tower/wiki) :)

ne0fhyk, 2016-07-21:
@kikislater This community shares freely when it comes to information, so if no one is answering, it's more likely that no one has experiences with this specific hardware.
In that case, you can contribute as well by testing, and reporting your findings through edit to the [Tower wiki](https://github.com/DroidPlanner/Tower/wiki) :)

In addition, the best place for this type of questions would be the [Tower forum](http://discuss.ardupilot.org/c/ground-control-software/tower).

kikislater, 2016-07-21:
Thank you for Tower forum. I think we are not able to edit wiki, we have to create a fork and push a pull request

ne0fhyk, 2016-07-21:
@kikislater The wiki should be editable by all github users without need to fork and submit a pull request. I just checked the settings to confirm it is. 
Can you try once more, and confirm you're able to edit/contribute to it.

kikislater, 2016-07-21:
Ok it's available for editing, sorry !

Genestealer, 2016-07-21:
I tried using EasyCap viewer and streaming to Tower however he's account viewer shuts down when it's sent to the background and tower is brought into the foreground in Android. I was hoping that we could add support for those devices to Tower.

kikislater, 2016-07-21:
So it's not working as easy as XYUSB chipset ! 
Did you tried with another device? Some android device like my huawei close background app

kushanshah11, 2016-07-21:
finally it is not supported if i use Easycap?

kikislater, 2016-07-21:
It should work if you are able to stream video to tower but not as plug and play as xyusb chipset

kushanshah11, 2016-07-22:
Have anyone know how to stream video using Easycap Device to Tower please let me know it would be helpful to me...

Thanks.

pedvi7, 2016-07-22:
I can say for sure that I've tried Easycap, for some reason it won't stream to the latest beta Tower with guiseco updates. It does stream ok to Easycap viewer app on my S6edge and tab 10.
I'm still planning on getting one off those Chinese generic UVCs guiseco used, but it's going to take about a month.
Any suggestions?

kushanshah11, 2016-07-26:
hi @pedvi7  i just want make one app that stream video using easycap device same as (Easycap viewer).
i search lot on it but didn't success.

if anyone have idea how to stream video from any RCA device please do needful.

Thanks.

kikislater, 2016-08-04:
May be in Qgroundcontrol, it should be possible to stream from easy cap