DroidPlanner/Tower #1761: Guided location sent with incorrect MAV_FRAME (using 0, should be 3)

rmackay9, 2016-06-16:
Copter-3.4-rc1 (and later) support different altitude frames (absolute, terrain, relative-to-home).  Tower always seems to send the altitude frame of "0" (MAV_FRAME_GLOBAL) which means the altitude is being specified as an alt-above-sea-level.  The smallest fix is to send MAV_FRAME_GLOBAL_RELATIVE_ALT (=3).

Search for "MAV_FRAME" on this web page: http://mavlink.org/messages/common

The best fix would be to stop using the MISSION_ITEM with a MAV_CMD_NAV_WAYPOINT inside and move to using the SET_POSITION_TARGET_GLOBAL_INT message and then use one of these supported mav-frames:
MAV_FRAME_GLOBAL_INT (=0)
MAV_FRAME_GLOBAL_RELATIVE_ALT (=3)
MAV_FRAME_GLOBAL_RELATIVE_ALT_INT (=6)
MAV_FRAME_GLOBAL_TERRAIN_ALT_INT (=10)

wqwdpxd, 2016-06-21:
This is actually a huge problem for guided mode. Needless to say I was very surprised when my hexacopter started rapidly loosing altitude after switching to guided mode on 3.4-rc1. Until this is fixed, the guided mode is pretty much useless. I mean, you could use it if you were fast enough to change the guided altitude to a couple hundred meters before the vehicle hits the ground...  Anyway, you did an awesome job explaining the causw of this and I wpuld really like to see a fix for this soon.

ne0fhyk, 2016-08-10:
@rmackay9 This issue is being fixed by pr https://github.com/dronekit/dronekit-android/pull/426.
For now I'm just using the smallest fix because the full fix requires the ability to detect the autopilot version (since it's only supported on 3.3+) which I'm still implementing.

closed 2016-08-10 by ne0fhyk

rmackay9, 2016-08-10:
@ne0fhyk, great, thanks!