DroidPlanner/Tower #1763: Tower closes after connecting to Solo?

quadquad, 2016-06-29:
Hi,

 My issue is that once I want to connect Tower to Solo to conduct pre planned missions the application connects, shows the "arm" and "dronie" icons on left down corner and informs that flight mode is loiter, but shuts down immediately soon after.

 Taken from Solo app -> Settings -> System Info
 App 2.3.0
 Solo 2.4.0
 Autopilot 2.0.20
 Controller 2.4.0
 Controller Firmware 1.2.11

 Taken from Tower app -> Settings
 Version Tower-v3.2.1
 Firmware Version -
 Vehicle Type Unknown

 3DR services is v1.5.3

 Hardware
 Samsung Galaxy Tab S2 (SM-T715) 8.0
 Android 5.0.2

ne0fhyk, 2016-06-29:
@quadquad did the issue occur with Solo v2.3.0?
Can you provide the Tower logs after the app crashes. You can find them under `/sdcard/Android/data/org.droidplanner.android/files/crash_log`. Pick the log file that matches the timestamp for the crash.

quadquad, 2016-06-30:
Here are the three examples,

[2016-06-29 13-00-12[1].txt](https://github.com/DroidPlanner/Tower/files/340755/2016-06-29.13-00-12.1.txt)
[2016-06-29 12-57-02[1].txt](https://github.com/DroidPlanner/Tower/files/340756/2016-06-29.12-57-02.1.txt)
[2016-06-29 12-56-27[1].txt](https://github.com/DroidPlanner/Tower/files/340757/2016-06-29.12-56-27.1.txt)

ne0fhyk, 2016-06-30:
@quadquad the logs you provided are pointing to an issue with the text to speech functionality. Try disabling it for now, and see if that solves the issue.

quadquad, 2016-06-30:
I disabled Google text to speech from 
"path: settings/apps/all. Scroll down to Google text-to-speech engine and click. Select disable."

 Voice assistant is also disabled, but under the accesability options I am not able to disable Samsung TTS module, and do not find it from list of all apps (or don't know what is the exact name of the module)

[2016-06-30 09-50-12[1].txt](https://github.com/DroidPlanner/Tower/files/340838/2016-06-30.09-50-12.1.txt)

ne0fhyk, 2016-06-30:
@quadquad I meant in the Tower app. You can disable speech output through `Settings` -> `User Interface` -> `Speech Output`

quadquad, 2016-06-30:
Under the speech output, the enable speech output box has been all the time unchecked

quadquad, 2016-06-30:
Problem solved, I deleted all pre installed unimportant applications and now it seems to work.

ne0fhyk, 2016-06-30:
@quadquad great to hear! Can you list the apps you uninstalled so we can attempt to pinpoint the issue if it occurs again?

closed 2016-07-20 by ne0fhyk