DroidPlanner/Tower #1767: TBS Crossfire - RC Failure Message

fingadar, 2016-07-11:
I tried to use the tower app on a Galaxy Tab with the TBS Crossfire, Firmware 1.34. The Crossfire recieves the mavlink telemetry. To connect them with Bluetooth was no problem, and so far, everything seems to work.
But there is one issue: The Tower app brings up an RC Failsafe error periodically. But because the pixhawk on board does not fall into failsafe mode and also there are no problems recorded in the log files, this seems to be an erratic failure message.
I'm quite shure, that the mavlink telemetry connection via Bluetooth is shortly interrupted and so the Tower app shows the "RC Failsafe" message. In this moment, values like voltage shown in the app are completly wrong. Then, just one or two second later or so, the mavlink seems to resync, and the Tower shows correct values again. This repeats about 10 times during a 20 minute flight.

Tower V. 3.2.1
Device: Samsung Galaxy Tab 4
FC: Pixhawk + Arducopter 3.3.3
Link to tlog: https://dl.dropboxusercontent.com/u/78165243/log_bluetooth_2016_07_10_20_47_18.zip

Please let me know if you need any other information.

sirPerna, 2016-07-20:
I fixed this from Crossfire side. Issue can be closed.

ne0fhyk, 2016-07-20:
@sirPerna Thanks for the update.

closed 2016-07-20 by ne0fhyk