DroidPlanner/Tower #1776: Survey - Save settings

0n3-70uch, 2016-07-26:
Hello,

is it possible to store the settings in survey mode? Like camera model, flight attitude and so on...

Thank you
Greetings
Fabian

ne0fhyk, 2016-07-26:
@0n3-70uch yes, in the `Editor` screen, click on the menu overflow buttons (three vertical dots in the screen top right corner) to open the menu.
There should be an option to export your current mission to a file.
To restore the saved settings, just used the same menu to import the previously saved mission back.

0n3-70uch, 2016-07-27:
@ne0fhyk Hi, thanks for your fast answere. But that is not what I mean. Is it possible to save the curent (most used) camera model and the parameters like flight attitude, direction, ... So when I make a new survey the parameters are the same like before. At the Moment I must check each parameter, that is really time-consuming.

Greetings Fabi

ne0fhyk, 2016-08-19:
Fixed by commit a947a7449fc61bf6779927794c4ef74a7a875f72

closed 2016-08-19 by ne0fhyk