DroidPlanner/Tower #1781: Adding unknown camera

flo-vienna, 2016-08-07:
I have just been trying to use a Canon S90 together with the android version 3.2.1 of Tower.
CHDK is running on the camera and I can trigger the camera via my RC (chan7) and via the mission planner software on my pc still the Tower software does not seem to trigger the camera.
I this some kind of issues with the S90 being missing from the list of cameras in the software ?

Why is the S90 missing and how can I properly add it ?

Thanks :)

TShapinsky, 2016-08-07:
If you go into mission planner (or a similar ground control station) you should be able to setup the camera trigger. In mission planner its in the camera gimbal section under initial setup and optional hardware

flo-vienna, 2016-08-07:
Yes, the camera is working in mission planner for windows (via software or RC), but it is not working when I try to use the tower android app.

flo-vienna, 2016-08-09:
Is there any way of debugging if the Android Tower app tries to trigger the camera ?

It seems triggering via the Mission Planner app seems to be very reliable but triggering via CH7 opt on the RC does (for unknown reasons) not seem to be 100% reliable.

flo-vienna, 2016-08-20:
Camera added

closed 2016-08-20 by flo-vienna